OctoText/OctoText #21: Installs but does not show up as avlable

jroman7, 2021-03-31:
Hello,

When i search the plug-in repository OctoText does not show up.  I can install via the URL though.  After i install it shows up in the disabled plugins.  I enable it but it does not show up on the left side so i can't make any changes.  After I restart OctoPrint The plug-in disables again.  

I have tried to install several times.  

OctoPrint 1.5.3

berrystephenw, 2021-03-31:
after a little investigation I'm betting that it's a python 2 distribution. If you follow these instructions and update your machine you should be all set:
https://community.octoprint.org/t/upgrade-your-octoprint-install-to-python-3/23973

jroman7, 2021-04-01:
Ahh  yes!  that fixed it.  Thank you!   Now i need to figure out how to use
gmail to send the text.  I keep getting a failed to send message.

berrystephenw, 2021-04-01:
great! Gmail is a pain but should be fixable. They require you to setup an app password. Try this and let me know if it works:
https://support.google.com/accounts/answer/185833?hl=en

berrystephenw, 2021-04-02:
closing this issue as it is not a bug.

closed 2021-04-02 by berrystephenw