OctoText/OctoText #30: request for SSL protocol

berrystephenw, 2021-04-01:
investigate the ability to use SSL protocol instead (or as an option) of TLS.

staffert, 2021-04-03:
I second that request.

berrystephenw, 2021-04-12:
> I second that request.

I have added this feature in the next release candidate RC5. Would you be interested in testing it? I can push it out either today or tomorrow.

staffert, 2021-04-12:
sign me up.

berrystephenw, 2021-04-23:
I believe this feature is working @staffert - if you can confirm this (I have one other person that is using this encryption and it is working for them) then I will close the issue.

wlbillmartin, 2021-04-23:
I went ahead and set up a hotmail account for the texting. That’s working great!
 
Sir William Louis "Bill" Martin Jr., ATP, Capt. (Ret.), OMS
Active Participant - Planet Earth
AKA:
William P. Lewis
Comedian/Radio Personality/Actor
http://www.williamplewis.com/
Phone: 949.933.6230
Fax: 909.510.8253

closed 2021-04-26 by berrystephenw