OctoText/OctoText #31: webcam flip doesn't show up in message

dave2492, 2021-04-01:
In my configuration, I enable both horizontal and vertical flip of the webcam image - so it turns out correctly in the OctoPrint display.  The image that is sent by OctoText however, is not flipped.  I am currently using email, not text.

berrystephenw, 2021-04-01:
Hi Dave. There is a known bug in the rotate/flip code. I'm actually surprised that you got an image at all! The code in main throws an exception because of bad parameters to ffmpeg. My fix needs testing, but it should be merged later tonight.

berrystephenw, 2021-04-02:
I've released a fix, please check it out and let me know if it is working for you.

dave2492, 2021-04-02:
I updated to 0.1.5 but I don't see a difference.  I get an image but it is upside down.  (Note the image in the message is what I would see in the control tab if I deselected the flip horizontal and flip vertical buttons in the webcam settings.)   An alternative for me would be to rework the webcam bracket to allow me to physically flip the webcam...

Perhaps I didn't have crash before since I am running a fairly recent OctoPi (0.18.0)?

berrystephenw, 2021-04-02:
Weird. If you can attach your logfile it should help me figure out what is going on. 

The bug that I was fixing was in my code - I wasn't passing the file name correctly to the program that flips the image. You might have a different issue.

dave2492, 2021-04-02:
[octoprint.log](https://github.com/berrystephenw/OctoText/files/6251036/octoprint.log)

I wonder if the flip options are effective via the URLs.  I notice that if I click on the test button for the stream or snapshot URLs in the webcam settings page, that I see the upside down images...

dave2492, 2021-04-02:
Found a post that the telegram plugin had a similar problem - might help here:
https://community.octoprint.org/t/global-webcam-orientation-settings-not-applied-to-telegram-plugin/23428

berrystephenw, 2021-04-02:
Hi Dave, I looked at the link, I think we are dealing with something different. The code explicitly looks for the rotation/flip settings and tries to modify the image accordingly. If you send the logs that would help debug the problem. Thanks!

dave2492, 2021-04-02:
I did post the octoprint.log file.  Do you want other(s)?

berrystephenw, 2021-04-02:
Sorry, that's my fault - I was reading your comment on my phone and didn't see the link. I took a look and I don't see any failures in the code. Are you using a USB camera?

dave2492, 2021-04-05:
Yes.  It is  a Logitech C260.  Due to the bracket I use, the camera is mounted upside down.

berrystephenw, 2021-04-05:
I have one of those on my ender 5. I'll see if I can duplicate the problem there.

berrystephenw, 2021-04-06:
I don't have a C260 - its a C270 and it is working fine. I'm able to rotate and flip the image. 
[image]
I'm at a loss to understand why yours is behaving differently !

berrystephenw, 2021-04-11:
closing as there is no further complaints of problems with webcam orientation.

closed 2021-04-11 by berrystephenw