OctoText/OctoText #36: Update from 0.1.4 to 0.1.5 in loop?

DangeMask, 2021-04-02:
I just installed the plugin and Octoprint keeps telling me "there is an update for OctoText".
In plugin manager, it show I have 0.1.4 and 0.1.5 is available, so I click Update now and it tells me, that I already have the newest version, and forces reinstall. Then the server restarts and the update notification shows again.

If I can find any more details for you to debug it, just tell me.

berrystephenw, 2021-04-02:
Thanks - I missed changing the version in setup! The loop should be fixed now. Please let me know if it is not.

DangeMask, 2021-04-02:
Unfortunately not. IMHO the release links to old commit instead of the new with correction. So octoprint keeps downloading the older file with wrong version number inside.
[image]
[image]

berrystephenw, 2021-04-02:
Full disclosure - I am not an expert at how the release process works with Octoprint! My understanding is that the URL that is used for the update (https://github.com/berrystephenw/OctoText/archive/refs/heads/main.zip) is always pointing to the code on that branch. I *thought*, and I have no doubt that it is incorrect, that the releases were used to initiate the push messages to Octoprint so that people would upgrade. In other words if I make a change to the main branch without a new release being generated - the next install would pick up that change anyway. I originally used tags where the release name was also the install zip name and that would mean every change would need a new URL. 
To test this I cloned the main branch into a new directory and it shows the new version:
[image]

maybe there is something else going on? can you upload the plugin install log?

berrystephenw, 2021-04-02:
Ok, scratch all that. It's doing the same thing on my development machine, you are probably correct. I'll work it out and get it fixed.

berrystephenw, 2021-04-02:
you are right, the release needs to be made after the changes are done. Pushing up another rev... sigh.

berrystephenw, 2021-04-02:
Ok, I think it's fixed. It should be at 0.1.6 and I haven't seen a second message to update again. Thanks for pointing this out. Please let me know if it is fixed for you.

DangeMask, 2021-04-02:
Yep, works ok now.
Sorry if my comment seemed arrogant, I was just trying to help find the point.
I use git almost daily, but only for version control, I never used release feature. It's new for me too.

I needed to wait for print to finish before testing the fix. And your plugin was useful for notification. Thank you for the plugin and for quick fix.

closed 2021-04-02 by DangeMask

berrystephenw, 2021-04-02:
no worries I didn't take it that way. There was clearly an issue and I'm new at all of this (I'm a hardware engineer) and trying to find my way. I'm glad you like it!