OctoText/OctoText #39: Won't install

d-portero, 2021-04-02:
I can't fiind Octotext from a search in the plugin manager. When I add it from the URL I get this error. 

Installing plugin from https://github.com/berrystephenw/OctoText/archive/refs/heads/main.zip...
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install file:///tmp/tmpjtQGKT/main.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /tmp/tmpjtQGKT/main.zip
ERROR: Package 'OctoPrint-OctoText' requires a different Python: 2.7.16 not in '>=3, <4'
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

I have tried this multiple times on different days and the same error happens.

berrystephenw, 2021-04-02:
Hello, This is happening because your distribution has an older version of Python installed. OctoText only works with Python 3 and above distributions. The good news is that you can upgrade to Python 3 (if your other plugins are compatible) by following the instructions on this blog post:
https://community.octoprint.org/t/upgrade-your-octoprint-install-to-python-3/23973
I hope this helps!

berrystephenw, 2021-04-03:
closing because it is a known limitation of the plugin.

closed 2021-04-03 by berrystephenw