OctoText/OctoText #47: Replace test button with toggle.

kyle7575, 2021-04-04:
Replace the test button, (Still accessable in settings) with toggle button.

Alternatively add an option to not send messages if under a certain print time so you are not bombarded with texts/emails during prototyping.

berrystephenw, 2021-04-05:
Hi Kyle, this is more of a feature request than a bug (issue). If you want to try out the latest code which has changes to the test button please see #34. 

Your second idea might be something similar to what another person requested, which was status that was reported on a time interval basis (set by the user). That is more complicated than it sounds, but I'm thinking about how to implement it.

closing this issue because it's not a bug.

closed 2021-04-05 by berrystephenw

kyle7575, 2021-04-05:
I apologize for submitting my request under the wrong tab, I'll be sure to
use the experimental branch with the new features.

berrystephenw, 2021-04-05:
no apology necessary, thanks for the feedback!