OctoText/OctoText #48: Octotext isn't triggered on filament runout

mpd352, 2021-04-06:
Stephen:

I tested the update.  Octotext was not triggered to send a text when the filament sensor detected no filament.

My log is attached.
[octoprint.log](https://github.com/berrystephenw/OctoText/files/6261913/octoprint.log)

berrystephenw, 2021-04-06:
Thanks for the log, that is helpful. I'm not sure why I'm not seeing the printer changing state, but it is clearly happening in the log. I'll keep investigating this.

berrystephenw, 2021-04-07:
Ok, I looked through your logs again - and there is a clue:
`2021-04-05 20:34:59,099 - octoprint.printer.standard.job - INFO - Print job paused - origin: local, path: Ender_Test.gcode, owner: mpd352, user: None
2021-04-05 20:34:59,107 - octoprint.plugin - ERROR - Error while calling plugin OctoText
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_OctoText/__init__.py", line 457, in on_event
    user = payload["user"]
KeyError: 'user'`

I'm not sure why there is no user keyword, but I'll be looking at this. Hopefully a fix coming soon.

berrystephenw, 2021-04-07:
I think I see what is going on. There isn't user because the pause is initiated by the plugin. Should be an easy fix, hopefully I have a change posted today or tomorrow.

berrystephenw, 2021-04-11:
closing this as it was fixed in RC3.

closed 2021-04-11 by berrystephenw