OctoText/OctoText #53: Error sending emails with Yahoo mail

lavik17, 2021-04-10:
The settings I used:
address: smtp.mail.yahoo.com
port: 465 (but I also tried 587)

I get this error:
```
Test message could not be sent
Exception while talking to your mail server, check your SMTP settings.
```
Could you please advise?

berrystephenw, 2021-04-10:
I'm pretty sure you should use 587 for the port, but if you have tried it then I would need to see your log files to determine if there was something else going on. The only thing in the logs that you might not want to share is your email address (the password is not exposed) so if you want to attach the log here that would help (feel free to sanitize it).

lavik17, 2021-04-10:
I will send the log file. Could you please tell me the location of the file?

berrystephenw, 2021-04-10:
It's in the settings->logging menu. click the download button next to octoprint.log
[image]

lavik17, 2021-04-11:
Thanks! Here is the log file. I've replaced my yahoo username with 'myusername' but otherwise nothing was changed.

[octoprint.log](https://github.com/berrystephenw/OctoText/files/6290992/octoprint.log)

berrystephenw, 2021-04-11:
weird. something is funny with the login to your server. I found a site which claimed the port should be:
      Yahoo Outgoing Mail Server (SMTP) - smtp.mail.yahoo.com (port 25)
Try that!

lavik17, 2021-04-11:
Tried port 25, the same error is displayed.
It seems that sometimes it takes some more time until the error message is displayed and Octoprint is unresponsive at that time (maybe 1 minute), but once the error message appears it goes back to function normally.

berrystephenw, 2021-04-11:
Ok -  I should have checked yahoo first. Check this link: https://ph.help.yahoo.com/kb/SLN4724.html
You can see that it _requires_ authentication. This isn't totally clear, but I think you have to generate an app password. Try the instructions here: https://ph.help.yahoo.com/kb/generate-manage-third-party-passwords-sln15241.html

[image]

lavik17, 2021-04-11:
That worked!
So for yahoo mail, you need the App password and use port 587.
An unrelated question - Could you add a feature where you select the destination email address?
e.g. I would like to use yahoo mail to send the email, but it should be sent to my Gmail address.

Thanks!

berrystephenw, 2021-04-11:
Already there! You just need to change the phone number to the user name, and the SMS gateway to the host address. For example if you want to send to OctoText@outlook.com you would do this:
[image]

lavik17, 2021-04-11:
Ah! Thank you, I guess I didn't pay enough attention to the small letters :)
All working now!

berrystephenw, 2021-04-15:
ok, closing this issue since it's solved.

closed 2021-04-15 by berrystephenw