OctoText/OctoText #54: Won't connect to gmail, twc, OR outlook smtps (no TLS?)

MattLogue, 2021-04-11:
The readme mentions nothing of encryption protocols supported. 

I tried the SMTP servers for all three services and got the general error that it failed to connect.

Bendermk6, 2021-04-11:
Mine works fine with gmail, all you have to do is to activate the option to allow less secure apps to log into your gmail.

https://support.google.com/accounts/answer/6010255

I have setup a additional gmail account for OctoText, because i have 2FA on my main gmail account and don't want to allow less secure apps to log into it.

And then the gmail smtp server:
SMTP gateway: smtp.gmail.com
SMTP Port: 587

berrystephenw, 2021-04-11:
OctoText uses TLS. If you are having trouble please use the discussion board, and read the FAQ which should help you for gmail in particular. See this post: https://github.com/berrystephenw/OctoText/discussions/1#discussion-3279932
The issues section on github is for bugs - closing as this looks like a problem with setup.

closed 2021-04-11 by berrystephenw

MattLogue, 2021-04-18:
Well still won't connect I tried Outlook... then back to google after turning off 2step auth, allowing less-secure. Just says authentication issue which could be 4 or 5 different problems.