OctoText/OctoText #55: Notification subject "Print was cancelled" if print was cancelled by user

Bendermk6, 2021-04-12:
I noticed that when i cancel a print by hand, the subject from the mail i receive is `Print Fail after 98765 seconds`
Aren't there different trigger for error/fail and cancelled?

Also the text itself can be improved with gcode-file-name and printer instead of 2 times the printer name:
`Print failed on: Prusa MK3S+ MMU2s cancelled Message sent from: Prusa MK3S+ MMU2s`

OctoText: 0.2.0rc4

berrystephenw, 2021-04-12:
Hi - there is a different message for Canceling vrs Fail. I don't check for a Cancel message - once I saw that a cancel generates an error as well there didn't seem to be any point. I'll play with it a bit and see if I get two messages on a cancel... 

Yes the messages can be cleaned up, thanks I'll fix it.

berrystephenw, 2021-04-14:
So I've looked at this and for some reason when you cancel a print, you get both messages. This seems redundant, but I've added the ability to choose which message you want to receive. I'll post this change in RC5 possibly today along with the other fixes and additions.

Bendermk6, 2021-04-15:
Cancel is working now on RC5

closed 2021-04-15 by berrystephenw