OctoText/OctoText #57: Mail is missing the "Date:" field

ojaksch, 2021-04-15:
Many thanks for this needful addon. Installed, configured and tested it yesterday and works like a charm.
But what about I was wondering about: One mailclient (Thunderbird) displays new mails from Octotext on the top ("new mail, today"), but another mailclient (KMail) says "there's a new mail, dated 01.01.1970". And showing at mail's source reveals that the **Date:** field is missing.
Would be nice if this is getting fixed.

berrystephenw, 2021-04-15:
Well that is interesting! I'll take a look and see what I can do.

berrystephenw, 2021-04-15:
Ok, easy fix. I'll push up a new release in a day or so. I want to give people a few days to try out RC5 before sending out another release. Once RC6 comes out it would be great if you could let me know if it works for you!

ojaksch, 2021-04-15:
Great, thank you. Will keep an eye on this and report back.

berrystephenw, 2021-04-23:
The Date: field has been added @ojaksch if you have had a chance to test the latest stable release please let me know if this has fixed your problem. Thanks in advance!

ojaksch, 2021-04-23:
Can confirm that fix is working and KMail is sorting the Mail to the top / today with the correct time and date.
Great, many thanks for that - and for your work too!

> Return-Path: <larry@example.org>
> Delivered-To: larry@example.org
> Received: from octoprint.example.org (octoprint.example.org [192.168.666.666])
> 	(using TLSv1.3 with cipher TLS_AES_256_GCM_SHA384 (256/256 bits)
> 	 key-exchange ECDHE (P-256) server-signature RSA-PSS (4096 bits) server-digest SHA256)
> 	(No client certificate requested)
> 	by mail (Postfix) with ESMTPSA id CF3B3207C1
> 	for <larry@example.org>; Fri, 23 Apr 2021 14:56:21 +0200 (CEST)
> Subject: : Test from the OctoText Plugin.
> From: larry@example.org
> To: larry@example.org
> Date: Fri, 23 Apr 2021 14:56:21 +0200
> MIME-Version: 1.0

closed 2021-04-23 by ojaksch

ojaksch, 2021-05-09:
Just to add a comment: Great tool. **Very** helpful. Printed a bicolor object today with a single-extruder machine and OctoText keeped me informed when it's about time to change the color/filament ("PauseAtHeight") while relaxing in the garden. "Hold my beer, honey, have to adjust the flux compressor..."
:satisfied: