OctoText/OctoText #58: Disable TLS?

mct75, 2021-04-16:
Hi there, I have what in 2021 is probably an edge case, I use plaintext SMTP over port 25 to a local SMTP server which then handles transit to Gmail over secure means.   My OctoPrint instance is inside an isolated firewall segment without internet access so I need to use the relay, but I never needed to use TLS with it so it's disabled.   Is there a quick change I can make to bypass the STARTTLS command?  

Relevant log output:

2021-04-16 08:34:17,329 - octoprint.plugins.OctoText - INFO - The test button was pressed...
2021-04-16 08:34:17,330 - octoprint.plugins.OctoText - INFO - request = <OctoPrintFlaskRequest 'http://192.168.1.59:5000/api/plugin/OctoText?%7B%22command%22:%22test%22,%22channel%22:null%7D' [GET]>
2021-04-16 08:34:17,330 - octoprint.plugins.OctoText - INFO - Sending text with image
2021-04-16 08:34:17,330 - octoprint.plugins.OctoText - INFO - Enable webcam setting False
2021-04-16 08:34:17,331 - octoprint.plugins.OctoText - INFO - 10.51.1.63
2021-04-16 08:34:17,331 - octoprint.plugins.OctoText - INFO - 25
2021-04-16 08:34:17,331 - octoprint.plugins.OctoText - INFO - before server smtplib
2021-04-16 08:34:17,370 - octoprint.plugins.OctoText - INFO - SMTP_server <smtplib.SMTP object at 0x7fe51405aa58>
2021-04-16 08:34:17,372 - octoprint.plugins.OctoText - ERROR - Exception while talking to your mail server STARTTLS extension not supported by server.
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/lib/python3.7/site-packages/octoprint_OctoText/__init__.py", line 145, in smtp_login_server
    error = SMTP_server.starttls()
  File "/usr/lib/python3.7/smtplib.py", line 752, in starttls
    "STARTTLS extension not supported by server.")
smtplib.SMTPNotSupportedError: STARTTLS extension not supported by server.
2021-04-16 08:34:17,372 - octoprint.plugins.OctoText - INFO - String returned from send_message_with_webcam SMTP

berrystephenw, 2021-04-16:
Hi, you didn't mention if you are using the main distribution or the RC candidate. For the 0.1.6 release there isn't an option to disable TLS. For the RC candidate (0.2.0rc5) you can use SSL for the connection. Neither of these distributions allow you to disable encryption.

Normally I don't mind adding features for people, but I think disabling SSL/TLS would cause more problems for people trying to setup their plugin (and me to support them). 

You could clone the distribution and comment out the line that starts TLS (line 166 in rc5) download the zip file and load it into Octoprint yourself... 
[image]

berrystephenw, 2021-04-23:
While I would love to add this for you, I do believe it will cause a lot of support problems. Marking this as won't fix and closing.

closed 2021-04-23 by berrystephenw

Digenis, 2026-09-21:
@dkrahmer, as the new maintainer,
do you also reject adding an option for plain SMTP?

Many users choose octoprint because they want to avoid the public cloud
so setting up a local mail server which is already behind VPN is a valid scenario
in which setting up TLS & signing up with a CA is pointless jumping through hoops.

If you do reject plain SMTP,
would you at least accept a PR adding an optional CA path for self signed certificates?