OctoText/OctoText #72: Message sent from: [blank]

DustyArmstrong, 2021-04-24:
Running OctoPrint on a docker container, when I receive emails from OctoText the "from" is blank.

`Message sent from:`

What does OctoText use to populate the from field? Hostname? My docker host has a hostname, but perhaps that isn't being translated through to OctoPrint. Is there an easy fix I could implement? 

Thanks.

berrystephenw, 2021-04-24:
Hi Dusty, that is kind of a bug since it should say something... but you should change the appearance name in settings:
[image]
That should put something in that field.

DustyArmstrong, 2021-04-25:
Great, thanks!

It now says `Message sent from: OctoText`, and the name put in the Title under Appearance shows up in the subject line. I'm fine with that, I mainly wanted it for email rule filing on keyword. 

I'll close this off.

closed 2021-04-25 by DustyArmstrong