OctoText/OctoText #74: Text messgae fails on on instance of OctoPrint

mpd352, 2021-04-29:
Berry:

OctoText is running successfully on 2 out of 3 of my OctoPrint servers.  The test message fails with a message in the log that the server connection unexpectedly closed.

This is my oldest instance of OctoPrint, and I may be missing a library I need and have forgotten about.  I attached the log.  

When you've got time, can you look at it and see if something makes sense?

Thanks,

Kevin
[octoprint.log](https://github.com/berrystephenw/OctoText/files/6396238/octoprint.log)

berrystephenw, 2021-04-29:
Hi Kevin, there is something very odd going on here. I see messages from octotext before the call to startup! Have you tried power cycling your Pi? It almost seems like there are two instances running...

berrystephenw, 2021-04-29:
Ok with the help of foosel - it looks like I'm getting a printer event that I'm not expecting on startup. Can you enable the logging for these events in the screen shot please? 
[image]

mpd352, 2021-04-29:
Will do, thanks.

Kevin Conde
Peoria, AZ

[image]<https://user-images.githubusercontent.com/44931130/116552647-4258b480-a8c7-11eb-8b31-98b11eea2d8a.png>

mpd352, 2021-04-29:
Here are all the logs after a a soft reboot, then a power off reboot.
[octoprint-logs.zip](https://github.com/berrystephenw/OctoText/files/6401672/octoprint-logs.zip)

berrystephenw, 2021-04-30:
It's very strange, the startup sequence is a little odd, but I don't see why it isn't working. The connection is being "unexpectedly closed" and that usually means there is a problem with the server login. Sometimes it means you have to create an app password like gmail or yahoo, or two factor authentication is turned on and it won't allow you to connect to the service. 
Have you logged into this account using your other machines with OctoText?

mpd352, 2021-04-30:
Berry:

Yes, I have two identical octoprint instances with all the same settings, and they are fine.  I did not need to create app passwords for them, and they work fine.

mpd352, 2021-04-30:
I fixed it, but I don't why I had to.  On the first two instances I manually entered my email pw just to test, and they both worked.  On the third, non-working, one I did the same.  It failed.  I went to my account and reset the password, then entered the new password into all three, and now they all work.  I'm baffled.

berrystephenw, 2021-04-30:
Ok, well I can’t explain that - but I’m glad you got it working.

Stephen Berry
Hardware Engineering Manager
GSSI
40 Simon St
Nashua, NH 03060

berrystephenw, 2021-04-30:
closing as it was some kind of setup or server config issue

closed 2021-04-30 by berrystephenw