OctoText/OctoText #84: Feature request: delay message

ojaksch, 2021-05-26:
While still enjoying OctoText and it's useful notifications, I'm missing a minor feature: a delay before sending the message.
OctoText is taking a photo immediately after the print jobis finished, but my photos are very blurred and jitter-ish as my camera is mounted at heatbeds edge and is still moving to it's home position when OctoText takes a snapshot. Hence it would be usful (for me), if I could set a delay of some seconds before OctoText's action.

[image]

Bendermk6, 2021-05-26:
You can just add a "G4 S2" to your end-gcode after the home movment.
The printer then waits 2 seconds before finishing the printjob.
https://marlinfw.org/docs/gcode/G004.html

I have something similar in my end-gcode for cooling down the hotend and shutting down the printer via my TP-Link smartplug.
```
G1 X0 Y200 F3000 ; home X axis
M84 ; disable motors
M106 S255 ; turn on fan at 100%
M109 R48 ; wait for hotend temp
M104 S0 ; turn off hotend
@TPLINKOFF 192.168.1.89 ; shutdown TP-LINK
M107 ; turn off fan
```

ojaksch, 2021-07-15:
Thanks @Bendermk6 ! I've tried this commando a few times at different positions in the end-gcode scriptbox, but it seems that my printer (Anycubic i3 Mega) ignores this. It answer "ok" when command is sent and that's all.
Never mind, I'm getting mails with photos of the finished print job and that's the main thing :)

closed 2021-07-15 by ojaksch

berrystephenw, 2021-07-15:
Sorry I never responded to this. I was kind of hoping the gcode trick would work!

I'm also trying to keep the settings page simple for the majority of people...