OctoText/OctoText #85: Feature Request:  Color Change Alert

jccbsl, 2021-05-29:
PrusaSlicer apparently sends an M600 command when the printer reaches a color change point (see screen capture).  It would be very useful to have this added to the Printer Event Options.
[image]

berrystephenw, 2021-05-30:
I'll take a look. It might not be too hard to do since I'm already parsing strings from the printer.

jccbsl, 2021-05-30:
Would a screen capture from the Terminal Window in Octoprint suffice?

jccbsl, 2021-05-30:
Actually, Octotext already sends a "Print Paused by System" message, so a color change alert message is probably irrelevant.  I will look at the log files and, if I see a difference in what is recorded, I'll pass it on.
Thanks!

berrystephenw, 2021-05-30:
Right - if the printer sends “echo:busy: paused for user” then you are all set, the plug-in already does what you want!

Stephen Berry
Hardware Engineering Manager
GSSI
40 Simon St
Nashua, NH 03060

jccbsl, 2021-05-30:
OctoText is destined to become one of my favorite plugins.  Thanks again!

closed 2021-06-04 by berrystephenw