OctoText/OctoText #86: Percentage not Correct

realbadfox, 2021-06-02:
I have activated the percentage communication, but it is apparently based on the number of layers and not on the runtime.
If the last layers have very short process, that is rather bad.
Would it be feasible to change this or to determine the calculation method yourself?

berrystephenw, 2021-06-02:
The percentage that is reported is the same as the progress bar in octoprint. It might be possible to process the gcode for each print - or hook into another plug-in that gives a good estimate of time. The first is beyond my current skill set, the second depends on the available plugins and if they have a way to get time info from them. 
I'll put it on my list of things to look at.

berrystephenw, 2021-06-12:
So I started working on this and there are a bunch of interesting problems with the implementation. The most difficult is that the estimated time can change during the print... I'm working on a solution but it will take some time.

berrystephenw, 2022-03-25:
I'm going to suggest that using the printtimegenius plugin might be a way to solve this issue. I haven't done much testing with the plugin but it is supposed to give much more accurate numbers for remaining time to print.