OctoText/OctoText #87: SSL Encryption failure

b-morgan, 2021-06-11:
I configured OctoText with my email credentials and checked the box to force SSL encryption. Sending a test message failed. I unchecked the box to force SSL encryption and a test message was sent successfully. See the attached log file for details (OctoText logging was set to DEBUG).

[octoprint.log](https://github.com/berrystephenw/OctoText/files/6641118/octoprint.log)

berrystephenw, 2021-06-12:
Hi - a couple of questions. When you force SSL encryption you should typically change the port number to 465, the log showed that it was set to 587, which is the default port for TLS. You could try changing the port and see what happens.
Also is this the same email server or a different one that worked without SSL? Typically an email provider only opens up one port for email (typically, but there are always exceptions).

b-morgan, 2021-06-12:
Same email server in both cases. Checking the box and changing the port fixed the problem. I was led astray by the error, [SSL: WRONG_VERSION_NUMBER] wrong version number (_ssl.c:1056). 

IMO, the description text is a bit confusing. Changing the order of the sentences and the formatting might help (see below) or it might be less confusing if there were two port inputs labeled SMTP Port (TLS) and SMTP Port (SSL) with defaults of 587 and 465. Then the check box for SSL Encryption only picks which input to use. 

For the Comcast / Xfinity email servers both 587 and 465 work. Just for fun, I tried 25 and that failed after a bit of time.

I'd change the order of the sentences by moving the sentence "TLS is the default protocol..." before the check box and aligned like the sentences under the SMTP gateway input. Then the check box and the sentence "Checking this box..."

Or you could just close this issue and leave everything alone. :-)

closed 2021-06-23 by berrystephenw