OctoText/OctoText #90: Feature Request: Helpers

jneilliii, 2021-06-13:
It might be nice to have helper functions in your plugin that corresponds to your API function that plugins could utilize to send emails/text with directly from python.

berrystephenw, 2021-06-13:
Interesting. So that would mean not going through the front end. Just out of curiosity what are the benefits? Speed?

jneilliii, 2021-06-13:
completely speed. and less processing. think of the round-robin that would have to take place if a plugin developer had to use the API...they would first need to import requests, submit the API request, wait for the server to respond, and then react whereas with a helper function you have direct access.

A perfect example of this use case it the helper functions in the MQTT plugin. It allows for other plugins to re-use it's code for broker connection, subscriptions, and what not.

berrystephenw, 2021-06-13:
Sounds good, it's new to me so it will take time for me to figure it out. I'm also adding another feature that I need to test. I'll put it on the list for the next release. Thanks!

berrystephenw, 2021-06-14:
Ok that was easier than I thought. I didn't implement a register/unregister interface, but the helper function is in there. I checked it into the rc branch, and tested it with https://github.com/berrystephenw/OctoPrint-Textapi (new checkin on that too). If you could take a look that would be appreciated. I still have a bunch of testing to do for the time based progress notification so I won't push the release for a while.

berrystephenw, 2022-03-23:
released new version of API - hopefully this satisfies this request. Closing this issue.