OctoText/OctoText #95: Mail "from" and SMTP Auth Credentials needs to be separated

csigristgit, 2021-07-11:
Some Mail Servers (my own also) needs 3 diffrent Informations to login. Mail from, Username and Password.  Username and Password are for SMTP Auth. 
Can you please add a Field for a separate Username?

Thanks :)

berrystephenw, 2021-07-11:
This is the way it works now. Are you saying that you need a different user name than the email address used to login to the SMTP server? The email address field is used to login to the SMTP server - it can be any string you want. It is also used in the From: field.  

If your server doesn't want to see the @hostname.com try not putting it in the form.

csigristgit, 2021-07-12:
I tried all possibilitys, but none of them works.
The Server needs to know a Username AND the associated mailadress. In some other Tools i need to type in this for example:

mail_server: (SMTP address)
mail_sender: ("from" address aka e-mail account address)
smtp_auth: (yes/no)
mail_username: (username for e-mail account)
mail_password: (password for  e-mail account)
mail_security: (TLS/SSL)
mail_port: (in my case 587)

if the "mail_sender" is empty or =username, the message will not be accepted...

Thanks for your Time :)

berrystephenw, 2021-07-12:
Let’s try debugging it with the logs before I modify the code. Can you set up octotext and enable debug logging and send octoprint.log please? The whole log is very helpful thanks.

Stephen Berry
Hardware Engineering Manager
GSSI
40 Simon St
Nashua, NH 03060

csigristgit, 2021-07-12:
Ok, after a little digging i maybe know, whats going on.
With only the username in the Email Address field, the Message was send by OctoText.
The Mailserver sadly blocked this Mail, but with a good Reason. 
The smtp Lib trys to find out, whats the sender address is.

In my case, the username is "prusaprinter". The smtp gateway is "smtp.abc.com".
The log from the Mailserver shows "server.abc.com" as the sender. This is wrong.
But i think the lib does it this way: The IP from "smtp.abc.com" is A.B.C.D. The MX entry in the DNS system for A.B.C.D is "server.abc.com". And this is the part after the @ . So the sender shows as "prusaprinter@server.abc.com". But it has to be "prusaprinter@abc.com". 

Long story short: the lib guess the Domain by himself, if no full Email address is typed in. Some Mailservers are not happy about that...

Thank you

berrystephenw, 2021-07-12:
Ok, it would still be nice to see logs. 

What I could do is separate the email address into two fields the way it is done for the destination address and then provide an option to send only the username for SMTP authentication. If I provide a separate username I worry that some mail servers won't check the email field (I think I can set the from: address to anything I want for Outlook) and octotext could be used to spam people... 

If this sounds like it would work, I'll start the changes on the RC branch and let you know when it is ready to test. You'll have to switch to release candidates for OctoText to get the update.

csigristgit, 2021-07-13:
Hello

Your recommendation sounds nice. I think this could work.
I'm at work right now. I will send you the logs in the evening.

Thanks

berrystephenw, 2021-07-14:
this is the settings checkbox that you will need to select to test this mode:
[image]
I'll publish the change on the rc branch and you should be able to pick it up when you switch to the release candidate:
[image]
I'll get the release out in a little bit. Look for release 0.3.2rc1

berrystephenw, 2021-07-15:
@csigristgit please let me know if you try this and if it works for you.

csigristgit, 2021-07-15:
Thanks. Sorry for my late response. I'm really busy right now. But i try to test this evening. worst case this weekend.

csigristgit, 2021-07-18:
So, I finally made it.
Install the Update, no Problems. The new fields as you described are there.
The Checkbox is checked and the Fields are filled with separate username and domain.
But the "From" Mailadress ist still wrong. 
I send you the Logs.

Can you check this?
Thank you :)

berrystephenw, 2021-07-27:
I've released a new version. Let me know if I was able to fix it this time!

csigristgit, 2021-07-28:
Thanks for your hard work :) 
I run a short test yesterday. It looks a lot better. But something is still strange.
Today or tomorrow i go deeper with my testing. I will report it to you.

csigristgit, 2021-07-29:
Hi Stephen

I tested it and found the issue. the option to use only the username is only for smtp auth.
but you used this also on line 441 " SMTP_server.sendmail(login, email_addr, msg.as_string())"
But it has to be "SMTP_server.sendmail(fromAddr, email_addr, msg.as_string())".
the "from" address is always the full mail address.

i changed the code at home and now it works perfect :)

do you think this is ok?

Thank you very much :)

berrystephenw, 2021-07-29:
Makes sense. I'll test it using fromAddr and make sure it didn't break anything else and release it again on the rc channel.

Glad it is working for you!

 Steve

closed 2021-08-30 by csigristgit