OctoText/OctoText #108: Feature request: Make elapsed time in email more human readable

surenz, 2021-12-06:
Hi there,

The plugin is awesome and removes all the hassle with set up of mail server and so on.
I would like however to get the time in the incoming mail (maybe in sms too) for Print done event in [days:]hours:minutes:seconds instead of seconds taken directly from the payload (as I see in the code).
Perhaps this information could be useful in the the other emails where elapsed time is present in the request (I have no much experience with the plugin yet).

Thanks.

berrystephenw, 2021-12-16:
The RC branch reports time in HH:MM:SS format - if you try that release let me know if that works for you. It will eventually be mainlined.

surenz, 2021-12-24:
After few prints finished successfully and few canceled (I don't receive mails for them though), it seems the functionality works correctly. 
Looking forward for your new version release.
Thanks.

closed 2021-12-24 by surenz