OctoText/OctoText #109: Message not including webcam snapshot since update

adam3654, 2021-12-19:
I have been using this plugin for a while now without issue, since the last update the text message I receive does not contain a snapshot. It does provide the 'webcam offline' image. My webcam is fully operational I can access it through TSD and Octoprint Connection tab. I looked through the settings and I don't see anything specific to path to webcam or anything that could be improperly configured.

berrystephenw, 2021-12-19:
Arghhh... I added a timeout to the webcam call but it was too short. I'm sending an update with a much larger timeout value. Please look for another update: 0.3.2rc4 momentarily and let me know if this fixes the problem for you.

adam3654, 2021-12-19:
Thanks! I'll install it ASAP, I'm mid print right now. 

Regards,
Adam

adam3654, 2021-12-20:
I was finally able to install the latest RC, and have confirmed with a test message that the snapshot is being sent. Thank you Steve!

closed 2021-12-20 by adam3654