OctoText/OctoText #111: 0.3.2rc4 webcam timeout .2secs still aggressive

Nazar78, 2021-12-30:
Taking snapshots from certain smartphones needs at least 0.5secs to wake them up, focus then take the shot. Older slower phones might require longer timeouts. Great if there's an option to specify this timeout?

berrystephenw, 2021-12-31:
Interesting, I haven't heard of people using their smart phones as a webcam. Here are my concerns:
* the initial problem was a hang because there was no response from the request, I would hate to introduce a setting that could hang the system. 
* I really wanted to keep the settings page simple. Adding more 'options' makes it more confusing. At this point I wonder if I should create an advanced page for settings that are uncommon - which is where I would put this option.

I'll ask around and see what kind of trouble changing the timeout could cause and get back to you.

berrystephenw, 2021-12-31:
sorry didn't mean to close it.

berrystephenw, 2021-12-31:
Ok, I contacted some of the developers that know this stuff well and I should be able to safely increase the timeout to 5 seconds. I'll push a new update soon. Please test rc5 and let me know if that solves your problem.

Nazar78, 2021-12-31:
Thanks for replying. I'm using my old Samsung Note 3 with [Octo4a](https://github.com/feelfreelinux/octo4a).

Perhaps we can have a simple webcam timeout in ms option just under the "Enable Webcam snapshots." or just put in an adequate value. From my tests, I've bumped up the timeout to 3000ms and it works fine.

Nazar78, 2021-12-31:
> Ok, I contacted some of the developers that know this stuff well and I should be able to safely increase the timeout to 5 seconds. I'll push a new update soon. Please test rc5 and let me know if that solves your problem.

5 seconds should do it, thank you!

closed 2021-12-31 by Nazar78