OctoText/OctoText #112: Feature Request: Send Email after backup

OllisGit, 2022-01-01:
Hi @berrystephenw,

happy new year and thanks a lot for this plugin!

I setup a new printer and now I want to backup my settings in a regular way. 
With the help of "Backup Scheduler" and "Google Drive Backup" plugin I can realise this. But instead of using a cloud storage, I would like to use my email inbox.

I looked into your "rc" version with the helper function (#90, very cool feature. I think I will change my current public apis as well. @jneilliii thx for the tipp), but unfortunately the provided ```send_text``` function is not so generic I need. It expect to send an image, but I need to send an zip file.
https://github.com/berrystephenw/OctoText/blob/rc/octoprint_OctoText/__init__.py#L419-L424

It would be really helpful, if you could provide something like this:

```
EMailBackupPlugin.py
...
emailMessage = EmailMessage()
emailMessage["Subject"] = "my Subject"
emailMessage["From"] = "OctoText@outlook.com"  # 'OctoText@outlook.com'
emailMessage["To"] = "ollisgit+octotext@gmail.com"
emailMessage["Date"] = formatdate(localtime=True)
emailMessage.set_content("This is my body message", charset="utf-8")

backupFileLocation = "/OctoPrint/data/backup/MEINER-backup-20211230-213621.zip"
fp = open(backupFileLocation, "rb")
emailMessage.add_attachment(
	fp.read(), maintype="application", subtype="zip", filename="myBackupFilename.zip"
)

# your helper function expect a EmailMessage or maybe better a MIMEPart as an argument
self.send_email(emailMessage)
```

In this way the plugin developer is responsible for defining the email-content. Your plugin is responsible for the connection-settings, the queueing and sending.

What do you think is the best approach to realise my requirements: adding new event "email after backup" to your plugin and/or make the helper method more generic, so that I can implement a EMailBackup-Plugin?

If you have an opinion of this, maybe I can provide you a pull request for this.

BR
Olli

jneilliii, 2022-01-01:
> I think I will change my current public apis as well.

I'd be careful completely replacing the APIs as I know of several external utilities that utilize them (ie OctoDash and OctoFarm).

berrystephenw, 2022-01-02:
It's a good idea, let me spend a little time looking at how I could implement this. Right now the API path uses the same queue as the main code and it might take a bit of restructuring to make this work.

berrystephenw, 2022-01-03:
Ok so I spent a little time looking at this, and it does cause me to rewrite a few things, but the main issue I think is that the plugin that calls OctoText is going to have to have the same knowledge of the To: address as OctoText already has. Unless you can reach into my plugin (another helper function?) you won't know the email address that is already set in OctoText. This seems like a lot of work when really all you are trying to do is send a zip file instead of an image. 

What I think would be easy to do is to change my API (since I don't think anyone is using it) to make the attachment more general and specify the filetype in the setup to the helper call. I'm thinking something like this:

```
        title = "This is the message title"  # text only, no special characters
        description = "this is the body of the message or email"
        printer_name = (
            self._identifier  # you can use this to inform people this is coming from your plugin
        )
        filename = (
            self._basefolder + "/static/img/backup.zip"
        )  # path to a file to be sent. Set to None if not used.
        fileType = ("zip") # would be .jpg for an image of course
        do_cam_snapshot = (
            False  # True tries to send an image from the webcam if enabled in OctoText
            # only one file is sent, either the specified file or webcam and the
            # file takes precedence
        )
        data = dict(
            [
                ("title", title),
                ("description", description),
                ("sender", printer_name),
                ("attachement", filename),
                ("fileType", fileType),
                ("send_image", do_cam_snapshot),
            ]
        )
        # using the helper function to call OctoText. send_text assigned in on_after_startup()
        if self.send_text:
            result = self.send_text(command="OctoText", data={"test": data})
            self._logger.debug(f"OctoText api result {result}")
            return flask.make_response(flask.jsonify(result=True, error=result))
```
All I should have to do is take the fileType and append it in my code instead of assuming it's a .jpg. 

There are also several fixes in place that affect how the email address is formed and how it is checked when logging into the SMTP server that would make it difficult to implement if the email was pre-formed before I received it. People have some crazy authentication setups (sometimes no password, sometimes only username etc) and they affect the EmailMessage itself.

Thoughts?

berrystephenw, 2022-01-03:
I decided to give it a shot, wasn't too difficult to implement. I verified that I can send zip files through the API, the code to do that is here:
https://github.com/berrystephenw/OctoPrint-Textapi
I need to do a little more testing to make sure I didn't break anything, but I think this will do what you want. If you want to try it the head of the RC branch has the checked in changes. I'll push a release once I'm happy its all working.

OllisGit, 2022-01-03:
Hi @berrystephenw,

thanks a lot for looking into this and also providing a first solution/idea. Looks like a solution for my current "backup-plugin".

I am not sure what you mean with "Knowledge of the To: address". Of course the Plugin-Author needs to know how to enter the receiver-email address format. 
But the default behaviour could be still provided by OctoText. It's up to the developer which fields of EMailMessage should be overwritten, if "To" is not defined, OctoText default is used.
I think the developer needs this flexibility. E.g. maybe he/she needs to set a different receiver or maybe multiple receivers or he wants to set CC/BCC. 

I am also started a PR, but as you already mentioned it is not so easy to provide a generic solution. The challenge is to improve the code with the pattern "separation-of-concern".... my PR tries the following (with few modification)

- strict separation of: sending the email (technical logic) and preparing/providing the needed data (business logic, like subject, body, ...)
- The message queue is the layer between business and technical logic.

Technical logic is responsible for the following stuff and shouldn't know anything about the email content/structure
- Connection handling
- Error-Handling and Retry-Mechanism

Business logic just provide the needed informations like from, to, subject, body and also the path of the image.

And then there is a single mediator-function that takes the provided business data and transform this into a message queue item which then is processed by the technical worker.

The public api function is the same like the mediator-function, it received the email-data (do some checks and enrich the email if needed) and put the data into the queue. 

Something like this, lets see how far I will get ;-)

BR
Olli

berrystephenw, 2022-01-03:
@OllisGit 

I appreciate that you are interested in helping with the plugin (or at least that's what it looks like here), but I'm not interested in major re-coding of the plug-in for this change. I'm sure that the code could use restructuring, but it has evolved over time to be flexible for all kinds of edge cases, and some of them involve specific formatting of the EmailMessage. I really don't want to fix the code for the sake of python general purpose purity, and break functionality that is working for people.  

Thanks for understanding!

OllisGit, 2022-01-04:
Hi @berrystephenw,

I totally understand your concern about changing a lot of stuff for only a new feature that only a few people will use in the future.

So take my PR #113 as an "inspiration" to improve your code (it is fully functional!). 
In the PR I dropped the idea with the default-values, it make the PR more unreadable. At the end, the implementation was relatively easy, because I could almost reused all of your code (only rearranging the code).

IMHO it helps a lot if the code is structured in technical/business logic parts. Makes it easier for testing and refactoring.

BR
Olli

berrystephenw, 2022-01-05:
Thanks Olli, I'll take a look at this when I have more time in the next couple of days. I appreciate the effort!

berrystephenw, 2022-01-05:
I'm still not sure how your application is going to know what to set the From and To entries - in some cases the From: address has to match the email account that is setup by the user (outlook checks), and the To: can be a text gateway that the user has setup. You can probably send the email to anywhere you want, but now you have two settings, one in Octotext and one in the external app.

OllisGit, 2022-01-06:
In the PR implementation there is no default assignment of fields like "from/to" of the email object. The developer is fully responsible to set all needed attributes. So he needs to "ask" the user via it's own settings-dialog.

But of course it's easy to implement something like this in OctoText:
```
receive_api_command

if (email_message["From"] is None)
        fromAddr = (
            self._settings.get(["username"]) + "@" + self._settings.get(["servername"])
        )
       email_message["From"] = fromAddr

...similar for "to" field ... 
```

berrystephenw, 2022-01-12:
Ok I've done some basic testing and merged your changes into the RC tree for RC6 release. I've tested the API with the code from [https://github.com/berrystephenw/OctoPrint-Textapi](https://github.com/berrystephenw/OctoPrint-Textapi). I've got a few more things to look at, but I'll plan to release it soon.

OllisGit, 2022-01-17:
Cool, thanks!
Now I can continue on my EMail-Backup-Plugin.

berrystephenw, 2022-01-20:
@OllisGit please let me know when you are satisfied that the API for OctoText meets your needs. I'll merge this branch with the master and release it. Thanks!

berrystephenw, 2022-03-22:
Implemented and released. Let me know if there are issues.

closed 2022-03-22 by berrystephenw