OctoText/OctoText #114: Chrome Autofill Overwrites Username and Password

jbyerline, 2022-01-07:
Everytime that I log into OctoPrint, the autofilled OctoPrint login from chrome will change the login credentials for my outlook account in the plugin settings. This causes the messages not to be sent because of invalid credentials. 

This could easily be fixed by changing the form type to not be of type username and password.

berrystephenw, 2022-01-10:
Sorry to hear that - I don't see this as a problem with the plugin though. If I change the form names so that browsers do not autofill the password then I lose the functionality of the password being hidden - which is a security feature I'd like to keep.

You should be able to tell Chrome to forget the password for Octotext, and even if it does autofill once you change it it should remember the new password. Why is the browser overwriting with another password? 

Last question - the only time the Octotext email password is changed is when the settings form is saved, are you doing this all the time? Maybe there is something else going on...

berrystephenw, 2022-03-25:
if you still have a problem and want me to take a look, let me know and I'll reopen this issue.

closed 2022-03-25 by berrystephenw