OctoText/OctoText #122: Multiple Emails / Phone Numbers

r0oland, 2022-03-17:
Would be great if one could send notifications to a larger number of mails / phones. 

Any chance you can include this? Or is that not possible due to SMTP restrictions or something like that?

berrystephenw, 2022-03-17:
Based on my initial search, it looks very doable. It's an implementation issue - I need to create a form that allows for multiple entries. I'll take a look at this soon!

r0oland, 2022-03-17:
Much appreciated, I know a little python myself, so let me know if I can help. But it seems more of a form/web issue right?

berrystephenw, 2022-03-17:
Formatting the email isn't too difficult - it's making the setup form easy to create a new email (jinja2 templates) and passing it to the code in a way that can be parsed. A text field would work, but might be ugly... I'll play with it and make sure it works before I go too far.  Thanks for the offer !

r0oland, 2022-03-21:
Sounds good! Keep it simple.
I'd need like no more than 5 but probably even less, so CC field sounds good!

berrystephenw, 2022-03-22:
Changes have been made to the rc branch. If you switch to the test branch in software update you can test it:
[image]

Bendermk6, 2022-03-26:
Hi Berry,

i just installed the RC and got smtp errors when sending test messages. After adding an adress to the cc field and try again it worked. Deleted it afterwards fom the cc and now its working without problems.

See attached log, maybe it helps.
I even deleted all data from the plugin and could recreated the error.
[octoprint (4).log](https://github.com/berrystephenw/OctoText/files/8355392/octoprint.4.log)

Chrees
Adam

berrystephenw, 2022-03-26:
Hi Adam, thanks for testing it. I think I know what this is.

Steve

Stephen Berry
Hardware Engineering Manager
GSSI
40 Simon St
Nashua, NH 03060

Bendermk6, 2022-03-26:
Oh, I just saw I got your name wrong, sorry Steve. ;)

berrystephenw, 2022-03-26:
No worries! Pushing a fix in a moment. I've also reorganized the settings page. Let me know what you think if you get a chance.

r0oland, 2022-03-27:
Sorry, only got around to testing it now, but works like charm! Thank you very much!

r0oland, 2022-03-27:
FYI: I have not tested multiple phone numbers.

closed 2022-03-27 by r0oland