OctoText/OctoText #123: Unable to send - Outlook or Gmail

Avrution, 2022-03-26:
I receive this error when trying either option:

2022-03-26 16:30:26,601 - octoprint.plugins.OctoText - ERROR - Exception while sending text, 'NoneType' object has no attribute 'replace'
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_OctoText/__init__.py", line 583, in on_api_get
    direct_send=True,
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_OctoText/__init__.py", line 359, in _prepare_email_message_and_send
    cc_set = cc_set.replace("\n", "")
AttributeError: 'NoneType' object has no attribute 'replace'

Gmail is setup with app password and outlook works elsewhere.

Ideas?

berrystephenw, 2022-03-26:
yes, my bad. You can temporarily fix it by entering something in the CC field and it should work as before. I'm pushing a fix in a minute.

Avrution, 2022-03-26:
Thanks - that fixed the issue.  Really missed not having an email plugin, so very glad to have found this.

closed 2022-03-26 by Avrution