OctoText/OctoText #135: Bug: "FROM" address is assumed to be the auth username & hostname

james-yoneda-ow, 2023-01-29:
Some SMTP servers (like sendgrid) use usernames for smtp authentication.  However, octotext assumes that the FROM address is always the auth username + password (or if the "use username" is checked, then only username).  Neither of these work in this use case.

# Example
* SMTP auth username is "api_user"
* SMTP auth password is "blah blah blah"
* Octotext sets the FROM as "api_user", which is not valid
* SMTP server (rightfully) rejects the message

# Suggestion
* Add a "FROM" box that's optional.  If it's empty, use the existing logic.  If it's filled, use this to override the FROM address.

berrystephenw, 2023-02-01:
Hi - I've been meaning to reply. Thanks for the concise bug report. At the moment I'm not currently working on changes to Octotext, and I'm not sure when I will be able to get to it.

However, I'm willing to review pull requests and do basic testing if you are willing to try your hand a modifications.

Thanks,
      Steve

csigristgit, 2023-03-12:
> Some SMTP servers (like sendgrid) use usernames for smtp authentication. However, octotext assumes that the FROM address is always the auth username + password (or if the "use username" is checked, [...]

Hi James
@james-yoneda-ow 
Are you usen the stable or the RC branch? I had the same Problem and got in contact with Steve. we figured a sollution out and put it in the RC. Can you try this and give a short feedback?

Greets, Chris