OctoText/OctoText #138: OctoText does not send any messages however test sending works

repcsi, 2023-03-29:
It seems that octotext does not hook onto events:
octoprint log:
2023-03-29 18:49:26,417 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2023-03-29 18:49:50,876 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 84614, 'printer_state': 'PRINTING'}
2023-03-29 19:02:27,724 - octoprint.util.comm - INFO - Finished in 31151.087 s.
2023-03-29 19:02:27,725 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Finishing"
2023-03-29 19:02:27,742 - octoprint.printer.standard.job - INFO - Print job done - origin: local, path: mk3_to_mk3+_upgrades/e-axis_mk3tomk3s_part1_02mm_petg_mk3_8h44m.gcode, owner: repcsi
2023-03-29 19:02:27,935 - octoprint.util.comm - INFO - Changing monitoring state from "Finishing" to "Operational"
2023-03-29 19:02:28,158 - octoprint.plugins.tracking - INFO - Sent tracking event print_done, payload: {'origin': 'local', 'file': '92ff5b2042b10b8b7686e1cc04dc2cb55f7af0e3', 'elapsed': 31151}
2023-03-29 19:04:26,419 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2023-03-29 19:04:50,850 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 85514, 'printer_state': 'OPERATIONAL'}

After the checking the source code I even tried adding attributes as I saw some exceptions in the log before:
config.yaml:
  OctoText:
    carrier_address: gmail.com
    en_printfail: Fail
    en_upload: false
    en_webcam: false
    phone_numb: my-name
    server_pass: account_pass
    servername: outlook.com
    show_navbar_button: false
    smtp_message: Prusa i3 event
    username: account_username
    smtp_name: smtp.office365.com
    smtp_port: 587

Send test:
[image]

Evident that I have job finish enabled:
[image]

What should I try next?

berrystephenw, 2023-03-29:
I'm not sure what you mean by hooking into events. It responds to the events it needs to.

There should be log events by Octotext on startup. If you aren't seeing them, make sure you have debug logging for Octotext turned on. There should be an event for every attempt at sending an email/text.

Let me know when you have logs with the octotext messages - I really need those to help.

      Steve

[Screenshot 2023-03-29 212041]<https://user-images.githubusercontent.com/7921375/228646538-28151f60-4a44-4a2d-8e8b-0eb4131b1512.png>

[Screenshot 2023-03-29 212132]<https://user-images.githubusercontent.com/7921375/228646549-bf3671a8-35ea-414d-ab40-fdd1c6320aae.png>

repcsi, 2023-03-29:
Now that I have enabled debug logging for octoprint.log I received the mail about the job start, is this how it's supposed to be?

berrystephenw, 2023-03-29:
logging is not required for the plugin to work. The logging functionality is part of Octoprint not the plugin.

Is it possible that you received an email from an earlier test?

repcsi, 2023-03-29:
No I don't think so as the job name is mentioned in the mail and I just prepared this one for testing... I will do some more tests tomorrow, wipe the logs and do a clean job run and I will supply the logs if the issue persist. Thanks Steve.

repcsi, 2023-04-03:
@berrystephenw I can't reproduce the issue because it is working fine since I set the octoprint logging to debug, and also after setting it back to info... So probably something was off on my end.

closed 2023-04-06 by repcsi