OctoText/OctoText #140: Webcam picture does not check for rotation settings.

ArdjanB, 2023-07-04:
I have Octotext on my Prusa MK3S. My webcam is mounted 'upside-down'. The settings in OctoPrint are set to rotate, and the live-view stream is OK, as well as the OctoLapse videos.
However, when OctoText sends an email with a picture, it is upside-down; it looks like OctoText does not use the rotational settings from OctoPrint...

berrystephenw, 2023-07-06:
Hi Ardjan,

It's difficult to figure out what might be happening without the debug log. If you could send that to me, I can take a look and see what might be going on. There is definately code to rotate the image if it is set that way.

Thanks,
      Steve

ArdjanB, 2023-07-06:
[image]
[image]
[octoprint.log](https://github.com/berrystephenw/OctoText/files/11967256/octoprint.log)

Thanks for looking into this!
I have uploaded the picture as it was sent with OctoText, a screenshot from the situation when the print was done (right side up), and the Octoprint.log, with the Octotext setting to DEBUG.

I noticed a depreciation warning about changes in the webcam handling starting with OctoPrint 1.9.0, but I cannot find so quickly where this affects me.