OctoText/OctoText #145: Not able to use external smtp with auth

radeksalomon, 2025-01-24:
Hi, I am not able to set the smtp on smtp2go to work with your pluging. Should it work?
mail.smtp2go.com
port 587
Checked: Check to use username only authentication. Rarely used option for mail servers that require a username only SMTP login.
Login and password filled.

Thanks a lot.
Radek

```
2025-01-24 18:43:09,167 - octoprint.plugins.OctoText - DEBUG - The test button was pressed...
2025-01-24 18:43:09,169 - octoprint.plugins.OctoText - DEBUG - request = <OctoPrintFlaskRequest 'http://10.24.24.31/api/plugin/OctoText?%7B%22command%22:%22test%22,%22channel%22:null%7D' [GET]>
2025-01-24 18:43:09,170 - octoprint.plugins.OctoText - DEBUG - Sending text with image
2025-01-24 18:43:09,175 - octoprint.plugins.OctoText - DEBUG - Preparing EMail 'Test from the OctoText Plugin.' and adding to Notification-Queue
2025-01-24 18:43:09,187 - octoprint.plugins.OctoText - DEBUG - Enable webcam setting False
2025-01-24 18:43:09,198 - octoprint.plugins.OctoText - DEBUG - Appearance name (subject): OctoText
2025-01-24 18:43:09,338 - octoprint.plugins.OctoText - DEBUG - mail.smtp2go.com
2025-01-24 18:43:09,338 - octoprint.plugins.OctoText - DEBUG - 587
2025-01-24 18:43:09,339 - octoprint.plugins.OctoText - DEBUG - before server smtplib
2025-01-24 18:43:10,228 - octoprint.plugins.OctoText - DEBUG - startttls() (220, b'TLS go ahead')
2025-01-24 18:43:10,228 - octoprint.plugins.OctoText - DEBUG - SMTP_server <smtplib.SMTP object at 0x5cb49aa8>
2025-01-24 18:43:10,229 - octoprint.plugins.OctoText - DEBUG - v3-octoprint-mk3s
2025-01-24 18:43:10,229 - octoprint.plugins.OctoText - DEBUG - Password supplied, attempting to log into mail server
2025-01-24 18:43:10,926 - octoprint.plugins.OctoText - ERROR - Exception while logging into SMTP server(send_email_message) (501, b'<v3-octoprint-mk3s>: sender address must contain a domain', 'v3-octoprint-mk3s')
Traceback (most recent call last):
  File "/home/radek-v3/oprint/lib/python3.9/site-packages/octoprint_OctoText/__init__.py", line 459, in _send_email_message
    SMTP_server.send_message(email_message)
  File "/usr/lib/python3.9/smtplib.py", line 970, in send_message
    return self.sendmail(from_addr, to_addrs, flatmsg, mail_options,
  File "/usr/lib/python3.9/smtplib.py", line 871, in sendmail
    raise SMTPSenderRefused(code, resp, from_addr)
smtplib.SMTPSenderRefused: (501, b'<v3-octoprint-mk3s>: sender address must contain a domain', 'v3-octoprint-mk3s')
2025-01-24 18:43:10,927 - octoprint.plugins.OctoText - DEBUG - String returned from send_message_with_webcam SENDM_E
```

jneilliii, 2025-01-24:
Based on the message I suspect this is because of the sender address being missing, and there's an open PR for this very specific thing, 

https://github.com/OctoText/OctoText/pull/136

radeksalomon, 2025-01-24:
> Based on the message I suspect this is because of the sender address being missing, and there's an open PR for this very specific thing,
> 
> [#136](https://github.com/OctoText/OctoText/pull/136)

OK. Thanks. I did not understood, that the checkbox affects also the From, not only auth.

Radek

closed 2025-01-24 by radeksalomon